scottchiefbaker/ESP-WebOTA #2: Use with WiFiManager.h

bill-orange, 2019-05-22:
@scottchiefbaker Can your library somehow be used with WiFiManager.h?  On an unattached ESP32 is would be good to update and setup Wifi without tethering to a PC.

Please advise.

P.S.  Nice work.  Worked on the first try!

scottchiefbaker, 2019-05-22:
Yes the two should work together just fine, but I haven't tried it. WebOTA binds to port 8080 by deault which should not conflict with WifiManager. 

I've never used WifiManager, so let me know your test results and I can update the docs.

bill-orange, 2019-05-22:
Will do.  I will let you know the result.

Invoke WifiManager before `init_wifi(ssid, password, host);`  ?

scottchiefbaker, 2019-05-22:
`init_wifi(ssid, password, host);` is just a wrapper around the regular joining WiFi stuff. I would anticipate that part would conflict with WifiManager?

For simplicity, connect to WiFi (however you do it) before running the OTA handle stuff. In my case I use `init_wifi()` because it simplifies the process of connecting to WiFi and starting mDNS.

bill-orange, 2019-05-23:
@scottchiefbaker  Understand. What would I need to change in your .h or .cpp to make `init_mdns (host)` available to my sketch? If I run it rather than `init_wifi()`  I think we are there.

scottchiefbaker, 2019-08-08:
Where did we leave this?

bill-orange, 2019-08-08:
I am running, but with a modification to WebOTS.h as described above.  I am hesitant to recommend WebOTA.h be revised. I don’t know what the impact of this change would in other people’s situations.

scottchiefbaker, 2020-01-28:
I just used WifiManager and WebOTA together in the same sketch with no changes required.

closed 2020-01-28 by scottchiefbaker