scottchiefbaker/ESP-WebOTA #3: ESP-01 ?

MoonMoon82, 2019-05-25:
Hi!
I'm using a ESP-01 board with a slightly modified usb-serial adapter like 
https://www.amazon.de/AZDelivery-ESP8266-ESP-01-USB-Adapter-Arduino/dp/B078J7LDLY
to upload sketches.

So usually I upload a new firmware by serial connection to a ESP-01 board. After I had some really good experiences with WebOTA on Wemos D1 boards, I also wanted to give it a try on a ESP-01 board. So I made this basic sketch:

`
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <WebOTA.h>

void setup() {
  delay(2000);
  pinMode(LED_BUILTIN, OUTPUT);
  WiFi.mode(WIFI_STA);
  WiFi.begin("*****", "*****"); //insert your wifi credentials
    while (WiFi.status() != WL_CONNECTED) {
      digitalWrite(LED_BUILTIN, HIGH);
      delay(100);
      digitalWrite(LED_BUILTIN, LOW);
      delay(100);
    }
    webota.init(8888, "/update");
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  webota.handle();
  delay(500);
  digitalWrite(LED_BUILTIN, LOW);
  webota.handle();
  delay(500);
}
`

Now I can access the webOTA interface on :8888/update and upload a new "firmware.bin" . Right after reaching 100% nothing happens anymore. I have to update by the serial interface to make the ESP-01 work again.

I watched the build output for the serial interface saying it's uploading a compressed file:

`Looking for upload port...
Auto-detected: COM7
Uploading .pioenvs\esp01_1m\firmware.bin
esptool.py v2.6
Serial port COM7
Connecting....
Chip is ESP8266EX

Features: WiFi
MAC: b4:e6:2d:29:46:eb
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Compressed 318688 bytes to 226708...
Wrote 318688 bytes (226708 compressed) at 0x00000000 in 28.0 seconds (effective 91.0 kbit/s)...
Hash of data verified.
 
Leaving...
Hard resetting via RTS pin...
 [SUCCESS] Took 33.90 seconds`

Could it be I have to upload a compressed firmware.bin ? I'm really sorry for beeing such a noob :(
Do you have a hint for me?

Kind regards MM82

MoonMoon82, 2019-06-29:
It's because the ESP-01 has not enough flash memory isn't it?

scottchiefbaker, 2019-06-29:
This library only supports the ESP32 and ESP8266.

MoonMoon82, 2019-06-29:
The ESP-01 is a ESP8266 board. It has 4Mbit flash on it. This isn't enough for a OTA update?

scottchiefbaker, 2020-01-28:
Did you ever figure this out?

scottchiefbaker, 2020-04-29:
Closing this due to lack of response. If it's still an issue let me know.

closed 2020-04-29 by scottchiefbaker