scottchiefbaker/ESP-WebOTA #5: A couple of warnings

AlexFundorin, 2019-08-13:
These are the ones I'm getting while compiling the sketch.
```
lib\ESP-WebOTA\src\WebOTA.cpp: In member function 'int WebOTA::handle()':
lib\ESP-WebOTA\src\WebOTA.cpp:61:14: warning: unused variable 'init_run' [-Wunused-variable]
  static bool init_run = false;
              ^
lib\ESP-WebOTA\src\WebOTA.cpp:72:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
lib\ESP-WebOTA\src\WebOTA.cpp: In member function 'int WebOTA::add_http_routes(WebServer*, const char*)':
lib\ESP-WebOTA\src\WebOTA.cpp:198:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
lib\ESP-WebOTA\src\WebOTA.cpp: In function 'int init_wifi(const char*, const char*, const char*)':
lib\ESP-WebOTA\src\WebOTA.cpp:256:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
```

scottchiefbaker, 2019-08-13:
Good catch... I just landed ca6c74b which should fix this. Want to test it out for me?

AlexFundorin, 2019-08-13:
Yep. The warnings are gone now.

closed 2019-08-13 by AlexFundorin

scottchiefbaker, 2019-08-13:
Thanks for the report. Otherwise is the module working for you?