scottchiefbaker/ESP-WebOTA #9: Having trouble compiling this

cems2, 2020-05-24:
Error is obscure so I'm unsure what might be wrong.  I'm trying to compile the Primes.ino example.
```
Users/cms/Documents/Arduino/libraries/ESP-WebOTA-master/src/WebOTA.cpp: In member function 'int WebOTA::init(unsigned int, const char*)':
/Users/cms/Documents/Arduino/libraries/ESP-WebOTA-master/src/WebOTA.cpp:40:22: error: no matching function for call to 'ESP8266WebServer::begin(const unsigned int&)'
  OTAServer.begin(port);
                      ^
/Users/cms/Documents/Arduino/libraries/ESP-WebOTA-master/src/WebOTA.cpp:40:22: note: candidate is:
In file included from /Users/cms/Documents/Arduino/libraries/ESP-WebOTA-master/src/WebOTA.h:4:0,
                 from /Users/cms/Documents/Arduino/libraries/ESP-WebOTA-master/src/WebOTA.cpp:5:
/Users/cms/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:69:8: note: void ESP8266WebServer::begin()
   void begin();
        ^
/Users/cms/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:69:8: note:   candidate expects 0 arguments, 1 provided
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```

scottchiefbaker, 2020-05-28:
I wonder if the underlying ESP8266 libraries have changed since I wrote this? I'll do some investigating on my own and let you know.

Can you provide the FQBN from the compilation output?

scottchiefbaker, 2020-05-28:
With the Arduino IDE 1.8.12 and ESP8266 library 2.7.1 I was able to compile the Primes example just fine. I selected NodeMCU version 1.0 as my board and compiled the prime example without errors.

Is your ESP8266 library up-to-date?

cems2, 2020-05-28:
I just checked my version numbers and my IDE was up to date but the ESP8266 library was old.  I just updated it but have not re-loaded your example today.  I suspect that was the problem entirely.

sorry to have bothered you without checking that first.

closed 2023-07-15 by scottchiefbaker