scottchiefbaker/ESP-WebOTA #11: Flash config wrong real: 1048576 IDE: 4194304

BozoTClown, 2020-08-19:
I have worked on this problem all day today. I have tried different IDE configurations, clearing flash, loading via different methods and non of them solved this issue. It was working fine and now it won't work on any of my devices. The only way I can get code to modify is to use esptool. If you have any suggestions, I am open to anything.

TIA
[image]
[image]
[image]

Stef-90, 2023-01-30:
I accidentally updated couple of devices with 1mb memory using esp12e board configuration in esphome it flashed but then It made all of them throw same error you showed here when i try to update them again i fixedthe files to use esp01-1m . Real 1m ide 4mb

I just compiled this ota only with arduino ide but cant flash it getting the same error. What was the trick?

BozoTClown, 2023-01-30:
If I remember correctly I had to erase memory before I could go any further. If tis doesn't help, let me know and I will dig deeper.

scottchiefbaker, 2025-07-11:
I'm going to close this as it's informational and not a "bug" per se.

I would love a Pull Request with documentation updates with this info. If you feel like submitting to the project.