scottchiefbaker/ESP-WebOTA #12: Cannot easily distinguish between WebOTA installations

jamesfowkes, 2020-09-19:
With several ESP32s running WebOTA on the same network, the only way to distinguish them is by IP address.

It would be useful to have a way to customise the HTML, such that the ESP32 can be identified uniquely by name.

There's lots of ways to do this, I wondered what your thoughts were about the idea, and the best way to implement it, if you think it would be useful.

scottchiefbaker, 2020-09-20:
I like this idea a lot actually. My first thought would just be to include the hostname in the HTML. No guarantee there IS a hostname though, and even if we were to display it, it may not help. 

We could add an optional variable to the `webota.init()` function to specify an identifying string?

I'm definitely open to adding this functionality, I just don't know what the "best" way is.

scottchiefbaker, 2023-07-15:
I included the MAC address in the HTML in 940d7bd48d927, but I have not published a new release with this commit (yet). Please try it.

closed 2025-07-11 by scottchiefbaker