scottchiefbaker/ESP-WebOTA #14: Crash when request "GET /webota"

AlbrechtL, 2021-02-12:
Hi,

Thanks for your great project!

I integrated your source code into my project https://github.com/AlbrechtL/BTT_TF_Cloud_AFW and I ran into the issue that the ESP8266 is crashing when the Web browser is requesting the page `GET /webota HTTP/1.1\r\n`.

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4024d985 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffce0 end: 3fffffc0 offset: 0190
3ffffe70:  00000001 00000001 3ffffea0 40212a1c  
3ffffe80:  4024d985 3ffffea0 3ffe942c 40212572  
3ffffe90:  3ffffef0 00000001 3fff1e2c 4020717b  
3ffffea0:  3fff1fac 001e001f 00fffed0 40212355  
3ffffeb0:  3ffffef0 3fff08c4 3fff1dec 401000e1  
3ffffec0:  3fff1dec 3fff08c4 3fff1dec 40204cb0  
3ffffed0:  62657700 0061746f 800069be 80ff0b7c  
3ffffee0:  3fff1dec 3fff08c4 3fff0884 40206d86  
3ffffef0:  6265772f 0061746f 87ff2400 0000005f  
3fffff00:  8000746f 4bc6a7f0 000069be 3fff0b7c  
3fffff10:  3fff08c4 00000000 4bc6a7f0 00000001  
3fffff20:  00000001 3fff1dec 40100462 1810624d  
3fffff30:  00000000 3fff08a8 3fff0884 3fff0b7c  
3fffff40:  00000001 3fff08a8 3fff0884 40206edf  
3fffff50:  3fffdad0 00000000 3fff0990 4020df31  
3fffff60:  007a1200 8548d0a7 3fff0500 4020c446  
3fffff70:  3fffdad0 00000000 3fff0530 40206f8b  
3fffff80:  3fffdad0 00000000 3fff0530 402012bc  
3fffff90:  3fffdad0 00000000 3fff0b3c 40100395  
3fffffa0:  3fffdad0 00000000 3fff0b3c 40213620  
3fffffb0:  feefeffe feefeffe 3ffe8584 401002d9  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

Do you have any idea what can be the problem?

AlbrechtL, 2021-02-12:
**Update**

I got a usable stack trace

```
Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
PC: 0x4020bd88
EXCVADDR: 0x4023d64f

Decoding stack results
0x40203cc4: FunctionRequestHandler::handle(ESP8266WebServer&, HTTPMethod, String) at /home/albrecht/snap/arduino/50/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WebServer/src/detail/RequestHandlersImpl.h line 37
0x4020931b: String::changeBuffer(unsigned int) at /home/albrecht/snap/arduino/50/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 156
0x40209367: String::reserve(unsigned int) at /home/albrecht/snap/arduino/50/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 146
0x4020931b: String::changeBuffer(unsigned int) at /home/albrecht/snap/arduino/50/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 156
0x40209399: String::copy(char const*, unsigned int) at /home/albrecht/snap/arduino/50/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 175
0x402093e6: String::String(char const*) at /home/albrecht/snap/arduino/50/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 36
0x4010020c: _umm_free at /home/albrecht/snap/arduino/50/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/umm_malloc/umm_malloc.c line 1295
0x402028ba: std::_Function_handler ::_M_invoke(const std::_Any_data &) at /home/albrecht/snap/arduino/50/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.h line 121
0x402092d8: String::invalidate() at /home/albrecht/snap/arduino/50/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 141
0x40203cba: std::function ::operator()() const at /home/albrecht/snap/arduino/50/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2465
0x40203cf6: FunctionRequestHandler::handle(ESP8266WebServer&, HTTPMethod, String) at /home/albrecht/snap/arduino/50/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WebServer/src/detail/RequestHandlersImpl.h line 44
0x402094c4: String::String(String const&) at /home/albrecht/snap/arduino/50/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/WString.cpp line 41
0x40203d7d: ESP8266WebServer::_handleRequest() at /home/albrecht/snap/arduino/50/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp line 590
0x40203118: FunctionRequestHandler::canHandle(HTTPMethod, String) at /home/albrecht/snap/arduino/50/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WebServer/src/detail/RequestHandlersImpl.h line 20
0x40203f7c: ESP8266WebServer::handleClient() at /home/albrecht/snap/arduino/50/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp line 303
0x401064ca: millis at /home/albrecht/snap/arduino/50/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/core_esp8266_wiring.c line 183
0x4020150f: delay at /home/albrecht/snap/arduino/50/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/core_esp8266_wiring.c line 51
0x40202ec2: WebOTA::delay(unsigned int) at /tmp/arduino_build_329829/sketch/WebOTA.cpp line 219
0x4020282a: loop() at /home/albrecht/src/ESP-WebOTA/examples/Blink/Blink.ino line 28
0x40209e30: loop_wrapper() at /home/albrecht/snap/arduino/50/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/core_esp8266_main.cpp line 125

```

AlbrechtL, 2021-02-12:
I found the problem!

According to https://arduino.stackexchange.com/questions/70391/esp8266-stops-working-when-i-use-2x-static-const-charprogmem the class `FlashHelper` must be used to copy the HTML code from flash to RAM.

Fix see PR https://github.com/scottchiefbaker/ESP-WebOTA/pull/15

scottchiefbaker, 2021-02-12:
I've never run in to this issue before... does the `fstr` definitely solve the problem for you?

AlbrechtL, 2021-02-13:
As far as I can see the issue is solved with `FPSTR`. I was able to do an OTA update of my ESP. But I didn't run extensive tests.

mlschuh, 2021-02-26:
I can confirm that PR #15 resolved this for me.

scottchiefbaker, 2021-02-26:
I haven't had this issue before, but I'm happy to merge it if it helps out.

closed 2021-02-26 by scottchiefbaker