scottchiefbaker/ESP-WebOTA #17: question about updating via curl - terminal

Mathteus, 2022-05-23:
I am using the lib 'ESP-Web OTA' in my project, and when I go to do the firmware update via curl through the linux terminal, I find the following error.

I add the following lines to init_wifi()

IPAddress local_ip(192,168,1,46);
IPAddress getway(168,168,1,1);
IPAddress subnet(255,255,0,0);
IPAddress primaryDNS(8,8,8,8);
IPAddress secondaryDNS(8,8,4,4);
if (!WiFi.config(local_ip, getway, subnet, primaryDNS, secondaryDNS)) ESP.restart();

to be able to get it by changing the ip of eps32.

using code in main

#include <WebOTA.h>
#include <Arduino.h>

void setup() {
   Serial.begin(115200);
   init_wifi("LAb", "test123", "esp-ota");
}

void loop() {
   webota.handle();
}

[image]

I did several tests and found that it only works if the ip of the esp32 is 192.168.1.* , when changing the ip of the esp32 does not work with the error 'Timeout for connection'.

I would like to know how to fix this in my code, if I'm doing something wrong.

scottchiefbaker, 2022-05-23:
That sounds like a local IP/Subnet issue on the sending box. Your ESP32 has a subnet of /16 (255.255.0.0), does your local box also? Are you able to ping the ESP32 when the curl fails?

Mathteus, 2022-05-23:
I tried to ping the ESP32 right after trying to update the firmware, and I got the following result.

[image]

About ESP32 having a subnet (255.255.0.0) I don't know if I understood it right, but I manually put it in the init_wifi() function, setting the variable, according to the subnet of my machine.

 IPAddress subnet(255,255,0,0);

scottchiefbaker, 2022-05-23:
Everything looks OK to me. I suspect the issue isn't with WebOTA, but instead with basic IP connectivity between your Linux machine, and your ESP32.

Mathteus, 2022-05-23:
could you tell why it only works with ip 192.168.1.* ?

Mathteus, 2022-05-23:
sorry, I clicked wrong and ended up closing the issues

scottchiefbaker, 2022-05-23:
I suspect the IP and subnet of your Linux machine, and your ESP32 are not the same.

Mathteus, 2022-05-23:
but if that were the cause then no ip would work.
or am i wrong about?

scottchiefbaker, 2025-07-11:
I've done updates with a different subnet, so I suspect this is something local to your environment. If this issue still persists please re-open this.

closed 2025-07-11 by scottchiefbaker