scottchiefbaker/ESP-WebOTA #18: Use WebOTA in FreeRTOS ?

Pigi-102, 2023-02-02:
That's not an issue but a question.
I've used your code in several projects and would like to use it also on a new one I've borrowed around ( an energy monitor for the house ).
Difference this time is that the project use FreeRTOS and works with tasks and not the usual loop.
I was wondering if you or someone has never thougth to use this code in a FreeRTOS environment.

Thanks for your code and thanks for your time :)

scottchiefbaker, 2023-07-15:
I have zero experience with FreeRTOS but I'd be open to pull requests.

ryancasler, 2023-08-28:
I know this is an old issue but I have done this through creating a task to initialize the webOTA that contains a while loop that runs the webota.handle method as well.  Something like this:

```
TaskHandle_t webota;
xTaskCreatePinnedToCore(handleWebOTA, "WebOTA", 10000, NULL, 1, &webota, 0);

void handleWebOTA(void* parameters){
  webota.init(8080, "/update");
  while(1){
    webota.handle();
  }
}
```

I thought I would run into heap problems since the task is not large enough to handle the whole sketch but it works just fine.  I think only because webota resets the board as soon as it is done updating.  I've been using this method for a while without issue.

Pigi-102, 2024-12-08:
> I know this is an old issue but I have done this through creating a task to initialize the webOTA that contains a while loop that runs the webota.handle method as well. Something like this:
>
> ``` [...]

I've tried your suggestions, but unfortunelly it does not works for me.
Probably because I'm on esp8266 and not esp32, but as soon as I add the lines, compile and upload, I start getting:

```Backtrace: 0x4008ce60:0x3ffbe170 0x4008d091:0x3ffbe190 0x400dd3b7:0x3ffbe1b0 0x400851a9:0x3ffbe1d0 0x400811fd:0x3ffc97c0 0x400812f8:0x3ffc97e0 0x400d1f2c:0x3ffc9800 0x400d1892:0x3ffc9830 0x40089575:0x3ffc9860

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
E (5252) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (5252) task_wdt:  - IDLE0 (CPU 0)
E (5252) task_wdt: Tasks currently running:
E (5252) task_wdt: CPU 0: WebOTA
```
Also if I move the webota.init from the task to the setup ( so it does not get executed in a task )  I get the same boot loop with the same message.

Any idea ?

Pierluigi

Pigi-102, 2024-12-09:
I found a workaround to the problem. Need to see if it will last long, but as for now it's working.
Problem is that in FreeRtos watchdog are enabled by default, and if some task goes longer than ( IIRC ) 10ms without leaving to other tasks a bit of CPU it will fire watchdog.
Reading comments on this : [https://github.com/espressif/arduino-esp32/issues/3871#issuecomment-913186206](url)   there is a workaround for tasks triggering a WD.

Long story short, you can avoid WD if the task get created like this:
```
TaskHandle_t webota;
xTaskCreatePinnedToCore(handleWebOTA, "WebOTA", 10000, NULL, tskIDLE_PRIORITY, &webota, 0);
```
with the priority ad tskIDLE_PRIORITY instead of 1 ( or more ).

This avoid the triggering of WD when the task is running.

So far is working perfectly. I will update in case of trouble.

scottchiefbaker, 2024-12-09:
I don't use FreeRTOS so if you find a solution I would love a PR will a documentation update for this.

Pigi-102, 2024-12-10:
> I don't use FreeRTOS so if you find a solution I would love a PR will a documentation update for this.

As I said, and thank to @ryancasler  for the first informations, it's quite easy.

In the sketch you add the usual ```#include``` as per a normal ( not FreeRTOS ) sketch, then , in the ``` setup() ``` function you add:
``` 
TaskHandle_t webota;
xTaskCreatePinnedToCore(handleWebOTA, "WebOTA", 10000, NULL, tskIDLE_PRIORITY, &webota, 0); 
``` 
for the task creation, pinned to core, and with the lowest priority possible, to avoid the watchdog to trigger, and finally you create the task code to be executed, as:
```
void handleWebOTA(void* parameters){
  // Put any configuration or customization here, as port, string for custom html and so on .
  webota.init(8080, "/update");
  while(1){
    webota.handle();
  }
}
``` 
eventually before the ```setup()``` funtion to avoid a declaration ( that's normal in C or C++ ) and you're done.

Hope it's clear, and it helps.

Pigi_102

ryancasler, 2025-09-06:
You only have one core on the ESP8266.  So, creating the task and creating the task pinned to  core are the same thing.  I suspect the difference in the way this is being handled between the esp8266 and the esp32 is the difference in the Arduino delay function.  On the 8266 delay, which is used in WebOTA, is blocking, preventing any other functions from running.  On the 32, the delay function is replaced with the FreeRTOS function vTaskDelay, which is non-blocking and therefore does not trigger the watchdog.  Anything that blocks is going to trip the watchdog timer on the 8266.  If you added a vTaskDelay call after your webota.handle, that would probably also prevent the WD timer from being tripped on the 8266.