scottchiefbaker/ESP-WebOTA #20: Identifier on web page

billbrach, 2023-07-14:
Not really and issue but an enhancement request:

Would it be possible to add some sort of identifier on the web page, so when the WebOTA web page is up, and about to load code to an  ESP board, you could identify the particular ESP board you are about to update ??  I have several ESP processors running in a system, and it would be very helpful, in the Arduino IDE, to pass an identifying name to that particular instance of WebOTA.  In this way, you would know you are about to update the correct board.  TIA !!

scottchiefbaker, 2023-07-15:
This is already implemented in 940d7bd48d927. I haven't published a new release with this commit in it (yet). Please try the code from this repo and see if it does what you need.

billbrach, 2025-07-11:
Thanks, will be a while before I can try this out.