scottchiefbaker/ESP-WebOTA #21: Examples won't compile properly with Arduino IDE 2.1.1

Dario503, 2023-07-18:
Get the following error when compiling examples or my own code (which compiled fine last year):

```
/private/var/folders/wv/bhszv4057tg00qff8cvwzvbr0000gn/T/.arduinoIDE-unsaved2023618-18874-z8cnao.s1xen/sketch_jul18b/sketch_jul18b.ino: In function 'void loop()':
/private/var/folders/wv/bhszv4057tg00qff8cvwzvbr0000gn/T/.arduinoIDE-unsaved2023618-18874-z8cnao.s1xen/sketch_jul18b/sketch_jul18b.ino:36:5: error: 'webota' was not declared in this scope
     webota.handle();
     ^~~~~~

exit status 1

Compilation error: 'webota' was not declared in this scope
```

Dario503, 2023-07-18:
Resolved by deleting the version downloaded by the Arduino IDE Library Manager (1.4) and installing the contents of the zip file found on Github.

So still a bug with what is getting pulled down as a library from within Arduino IDE.

I'm on an Apple Silicon Mac BTW. Arduino IDE v2.1.1

scottchiefbaker, 2025-07-11:
Now that the Arduino version is updated this should not be an issue.