scottchiefbaker/ESP-WebOTA #22: Interrupt block update?

netmindz, 2023-10-21:
Do interrupts block the update? It looks like they do

Could you add `noInterrupts` and `interrupts` to your code or provide some form of callback like the ArduinoOTA.onStart ?

netmindz, 2023-10-21:
Firmware update initiated: firmware.bin

assert failed: xQueueSemaphoreTake queue.c:1554 (!( ( xTaskGetSchedulerState() == ( ( BaseType_t ) 0 ) ) && ( xTicksToWait != 0 ) ))

Backtrace: 0x40083785:0x3ffbf2fc |<-CORRUPTED

ELF file SHA256: f4d9d1357296968f

netmindz, 2023-10-21:
Callback would be preferable as I've found detaching and attaching my own interrupts again looks to be better than disabling all.

As a workaround I'm detaching and then attaching either side of the `webota.handle();` but that's not ideal it's removing even when we aren't actually doing an updating

netmindz, 2023-10-21:
See code suggestion https://github.com/scottchiefbaker/ESP-WebOTA/pull/23

scottchiefbaker, 2023-10-23:
I have never used interrupts in my Arduino code so I didn't know this was a issue. Is the issue then that an interrupt fires while we're in the middle of uploading an interrupt and that breaks the update?

Seems like a simple `noInterrupts()` while were actively updating would be a simple solution? Looking at your callback solution it seems full featured and perhaps a little overkill. This project has been public for several years now and this hasn't come up before. I'm leaning towards the more simple `noInterrupts()` solution at the moment unless you can convince me otherwise.

netmindz, 2023-10-23:
Yeah the update gets only to 10% in my case before the interrupt kills it.

Disabling all interrupts didn't actually work for me on the ESP32.

As for the the complexity, registering the callbacks is entirely optional so doesn't make it any harder to use for users who don't want that. By using callbacks users can also do things like change a status LED to indicate update in process. 

The structure is lifted from AuduinoOTA  so well be familiar to users

scottchiefbaker, 2023-11-09:
Is there any reason **not** to call `noInterrupts()` inside of the HTTP upload section of the code? I cannot think of a scenario where you would WANT to break off your upload mid-stream to handle an interrupt.

I'm thinking it's a no-brainer to add `noInterrupts()` to the code, and then also look at adding the callbacks? What do you think?

netmindz, 2023-11-10:
Initially I tried using noInterrupts() in my callback but that didn't work so I swapped to detaching my rapidly firing interrupt. I'm didn't investigate why it didn't work, but it's possible that interrupts are needed for WiFi or something else that's part of the overall stack on an ESP32

netmindz, 2023-11-10:
https://forum.arduino.cc/t/enabling-wifi-conflicts-with-button-interrupts-resolved/644348/7

scottchiefbaker, 2023-11-10:
Does `noInterrupts()` just not doing anything on ESP32? 

Either way I'm leaning towards using it inside of the actual update code as it can't really hurt?

netmindz, 2023-11-10:
I need to tripple check, but I believe that disabling all interrupts will stop the WiFi stack working

scottchiefbaker, 2025-07-11:
Were you able to test `noInterrupts()` with WiFi? I'd like to land this.

netmindz, 2025-09-04:
Not had time to revisit this project sorry, but should be easy enough for you to test WiFi and noInterrupts() in your own test setup